PetRover/board #2: Add power pad to TPS55340

brycedcarter, 2015-09-12:
[image]

nnewhart, 2015-09-12:
I remember seeing this when I was making the footprint, but wasn't sure what to connect the pad to? Is it not connected to anything? or is it exposed ground?

brycedcarter, 2015-09-12:
yes, it should be connected to AGND... but that can happen during layout. For the footprint itself, you just need to make a solder mask defined pad where the underlying copper is 5X3.4 and the exposed copper is 2.31X2.64

brycedcarter, 2015-09-12:
[here](http://eagle.support.eng.narkive.com/FxgvYbra/how-do-i-do-a-solder-mask-defined-pad-with-eagle) is a link for how to do solder mask defined pads

brycedcarter, 2015-09-19:
Done
[image]

closed 2015-09-19 by brycedcarter