PetRover/board #12: STOQ023251 not wide enough

ajbezreh, 2015-09-17:
New to this software, but I think the center of the pin to the outside of the package should be 2.6mm, and it's far less than that. 
[image]

In the image below, the grid is set to 2.6mm (my way of showing that it's not wide enough?
[image]

ajbezreh, 2015-09-17:
Also, does it need to be 9.2mm in the other direction? It's currently 2.4mm. 
[image]
Not sure if these dimensions mattered, but I figured it might change spacings on the board if they're actually different.

ajbezreh, 2015-09-17:
If it is an issue, then it's also an issue with the 4 pin connector
[image]

brycedcarter, 2015-09-20:
fixed

closed 2015-09-20 by brycedcarter